ubc-subbots/triton #9: Write test suite for pipeline_manager

loganfillo, 2020-10-09:
We need to write a test suite for the pipeline manager to ensure it acts correctly.

Over-TheRainbow, 2020-10-19:
What should I check for:
"If pipeline_manager node configure pipeline service is used with a pipeline type that does exist, the yaml is well formatted, but does not contain the pipeline paramter namespace, the service call should be unsuccessful"?

The pipeline manager can create a warning 'Pipeline config YAML needs pipeline namespace', but it does not check do a specific test.
That message is currently supposed to appear if pipeline_params is None. Should I instead create a different message there, and after loading params try to check if namespace was loaded?

loganfillo, 2020-10-19:
@Over-TheRainbow  If you look at one of the config files, under `ros__parameters:` there should be a line `pipeline:` which is a namespace for the parameters needed for the pipline. If this `pipeline:` didn't exist then the yaml parser would not be able to find the pipeline parameters and pipeline_params would be None. 

One more thing, I think to avoid clutter in the PipelineType.msg, we should just use the "example" type as the type in all the test cases,  and just use the config file name param to specify which yaml the test should be using. For the missing yaml case, just use a config file name that doesn't exist and it should execute the same logic. Also if you end up reusing most of the code for each test case, it might help to encapsulate that in a helper function to avoid repeatability

Over-TheRainbow, 2020-10-30:
For 
"If pipeline_manager node configure pipeline service is used with a pipeline type that does exist, the yaml is well formatted with pipeline namespace, but one of parameters has an invalid type, the service call should be unsuccessful",
do you know if there is a log message that is printed already for this case? If there isn't, how should I compare the parameters to to check their type? Should I just check if they are strings / lists of strings according to the where the parameters are declared in the constructor?

loganfillo, 2020-10-30:
Weird, it looks like it hangs when trying to set a parameter of an incorrect type. I suggest the following; in _create_param_object, when you get the param_type from the curr_val, also get the type of the value to set (param_val), and make sure these types are equal, otherwise throw an exception or change the logic so that _load_params_from_yaml will return false in this case. Make sure to also print a message when this happens. Check out the [source code for  rclpy parmeters](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/parameter.py) for further understanding of how parametrs work.

Over-TheRainbow, 2020-11-08:
I'm a bit confused how to test the _load_component function to not load a component successfully. I tried using a test yaml with two components with the same name, but it was successfully loaded, though it did complain a bit. Not sure how to get it to be unsuccessful, do you have anything that I could try?

loganfillo, 2020-11-12:
@Over-TheRainbow  Sorry I just saw this. You should be able to load the same component multiple times (i.e load two example::ComponentOne nodes) but right now the node_name is static so it will be the same and that's why it complains. The main cause of the load_components not working is if you give it a component that simply doesn't exist (i.e example::DoesNotExist). The point in the code where you should see that the load_component failed is where we check "if not res.success", then log the warning. Basically what we want to do is make sure that the warning is printed when a config file contains a component that isn't in the corresponding package, or a package doesn't exist. An example config file for the first case could look like 

pipeline_manager:
  ros__parameters:
    pipeline:
      components:
        - example::DoesNotExist # this component is not in the package
      pkg_names: 
        - triton_example
      namespace: /triton/example
      remap_rules:
        - /triton/example/component_two/in:=/triton/example/component_one/out
    use_sim_time: false

and for the second case
pipeline_manager:
  ros__parameters:
    pipeline:
      components:
        - example::ComponentOne
      pkg_names: 
        - triton_does_not_exist   # package doesn't exist
      namespace: /triton/example
      remap_rules:
        - /triton/example/component_two/in:=/triton/example/component_one/out
    use_sim_time: false

Over-TheRainbow, 2020-11-13:
I just wanted to check if the output for your second case is correct. I was surprised to see that it aborted with a message that it failed to load **example::ComponentOne**, but it makes sense to me that it couldn't load the component because it had a non-existent package. This should be good output right? Or should it be changed?

loganfillo, 2020-11-14:
@Over-TheRainbow I am not sure exactly what the message should be, but to find out, use the CLI component interface to do some test commands and see what they produces. Behind the scenes, the CLI is calling the same service we are: https://index.ros.org/doc/ros2/Tutorials/Composition/

PR #54 (2020-11-20): Andrew/issue9
closes #9
closed 2020-11-20 by loganfillo