ubc-subbots/triton #37: Make a state_maintainer node which can be launched in the pipeline and maintain a state

loganfillo, 2020-10-28:


Over-TheRainbow, 2021-01-24:
I was able to set the pose the node uses using the config yaml, and convert the roll pitch yaw into a quaternion. Is there anything else I need to do or should I just start cleaning this up?

loganfillo, 2021-01-26:
> I was able to set the pose the node uses using the config yaml, and convert the roll pitch yaw into a quaternion. Is there anything else I need to do or should I just start cleaning this up?

@Over-TheRainbow 
The next step would be getting it to run in the pipeline, this would be easy for the simple no-offset case, but the case where we need to add parameters if we want an offset will be tricky. Try looking into the LoadNode service becuase I know it has a parameters argument so that you can load a node with parameters. We will have to adjust the pipeline config file format a bit to allow for passing parameters in the config which will be passed to LoadNode

closed 2021-02-27 by Over-TheRainbow