ubc-subbots/triton #56: Create a system for generation of underwater dataset

loganfillo, 2020-11-21:
We need to generate a dataset for object recognition given a rectangular planar object. This is how the generator should work. 

1. We have our planar object at the origin, and facing the x axis
2. We have an array of N camera forms a semi sphere around the object.
3. The camera array will move from it's starting pose along it's X axis (either forwards or backwards) to it's end pose which will be a smaller/larger semi sphere
4.Each camera will have an associated gazebo plugin which determine the bounding box of the object at a certain timestep. This plugin will need the dimensions of the object because it needs to know the points of the rectangular plane that the object exists in.
5. Each camera will also have M underwater_camera nodes associated to it, one for each water type
6. For each camera, there will also be a synchronised_image_saver node which subscribes to the M underwater_camera nodes associated to the camera, and also the bounding box plugin associated to the camera. This node will save the underwater_camera images and the associated .txt file which contains bounding box information (for yolo to train on)

kevinh42, 2020-12-20:
Bounding Box plugin has been created which can be loaded in a world (with adjustable location/size of the 3D bounding box) and can publish ROS messages.

https://github.com/ubc-subbots/triton/tree/kevin/bounding-box-plugin

kevinh42, 2021-01-17:
I added a Python node to get the output of the underwater camera and the bounding box plugin and save the data with labels.

PR #62 (2021-02-14): Kevin/bounding box plugin
closes #56
closed 2021-02-14 by loganfillo